gen2brain/cam2ip #1: Memory accumulation (need streaming support)

MintyOwl, 2017-10-04:
Great project.

However, running on browser without streaming support will exhaust memory pretty soon. Its great if you only want to peek every now and then, but to keep it live for a long time is not an option.

MintyOwl, 2017-10-04:
closing it since, the primary cause of memory acc. may be the debug tool

closed 2017-10-04 by MintyOwl

gen2brain, 2017-10-04:
Hey, thanks. I was thinking about some buffer, configurable, lets say 5M, and something like this https://github.com/opennota/h264 , so I can create video from images, and play that instead. But still thinking what is possible and what are the options.

MintyOwl, 2017-10-04:
I tested further on Microsoft Edge and the browser seems to not bloat the RSS, which is great, so I like the current format. However, H264 can definitely open up many other possibilities.