gen2brain/cam2ip #4: fatal error: 'jpeglib.h' file not found

montanaflynn, 2018-01-29:
On a mac:

```
../../antonini/golibjpegturbo/decode.go:32:10: fatal error: 'jpeglib.h' file not found
#include <jpeglib.h>
```

gen2brain, 2018-01-29:
You need to install lib/headers for turbo-jpeg, probably you can do it with brew on macOS.

Or, you can uncomment this line https://github.com/gen2brain/cam2ip/blob/master/encoder/encode.go#L6 and comment golibjpegturbo, but native image/jpeg is noticeable slower and takes more CPU.

montanaflynn, 2018-01-29:
Yes I had to install that requirement with [homebrew](https://brew.sh/):

```
brew install jpeg-turbo
```

I think you should list this in the README under the requirements section.

gen2brain, 2018-01-29:
Thanks, I added libjpeg-turbo to requirements.

closed 2018-01-29 by gen2brain