gen2brain/cam2ip #5: OpenCV 3 Support

r0l1, 2018-03-09:
Would be great if cam2ip would support the current OpenCV releases.

r0l1, 2018-03-09:
https://github.com/hybridgroup/gocv

gen2brain, 2018-03-09:
Yes, I also wanted to add support for latest OpenCV releases. Some reasons I preferred ver 2 is because it is in C not C++, and it has support for Android native method to access camera (but up to 4.x, not possible in later versions), now it must be used only with Android Java project over JNI. 

OpenCV 2.x can be left in repo, and can be available with some tag, don't need to be dropped.

r0l1, 2018-03-10:
Didn't know about the limitations with Android. I like the idea of multiple backends selectable through tags.

By the way: Great work with cam2ip.

gen2brain, 2018-03-10:
Waiting for https://github.com/hybridgroup/gocv/pull/125 , I need it accepted before I can add support for ver 3. Not possible to add similar function to cam2ip because I also need Step() function exposed.

gen2brain, 2018-03-14:
I added support for ver 3 in https://github.com/gen2brain/cam2ip/commit/3c0c949f314ee56f20871ab99c04ace9510cc80a . For now available if `-tags cv3` is used. PR at gocv is accepted but for now it is in dev branch.

gen2brain, 2018-04-17:
README is updated, for now default is ver 2.x, cam2ip uses less memory with it. For 3.x just pass `-tags cv3` (and install gocv before that, it is still not go gettable).

closed 2018-04-17 by gen2brain