gen2brain/cam2ip #6: Raspberry Pi version 2 executable are broken

rhoppel, 2018-03-13:
Installed "cam2ip" on a Raspberry Pi version 2 and got the following when executing.
~```
/bin $ cam2ip
cam2ip: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by cam2ip)
cam2ip: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by cam2ip)
```

gen2brain, 2018-03-13:
Installed like you used binary from releases? That one is cross compiled, probably with newer glibc then you have on system. But that CXXABI looks strange, probably since I added video support, that one adds ffmpeg etc. 

You can always just compile on real device, not sure what version of OpenCV is available, ver3 is ready but still waiting for new release of bindings https://github.com/gen2brain/cam2ip/issues/5 .

rhoppel, 2018-03-13:
the version on Raspberry Pi v3 works great.    Well done.   System resources are very low compared to other solutions.   I would like to use it on my older Raspberry Pi platforms to turn them into dedicated USB camera hosts.  I am curious to see how the can be used with multiple camera.

gen2brain, 2018-03-13:
Hmm, I deleted duplicate comment from github and now both are gone, I guess you received comment via email. Binaries are cross compiled on my system, I have toolchains installed with Gentoo crossdev. I probably installed that with newer gcc/glibc etc. not sure what exactly is problem. 

You should be able to install go, opencv and jpegturbo on real device, and install there, for test, I will see when I find time what exactly is the problem, will have to find and install system on old RPi I have somewhere.

rhoppel, 2018-03-13:
I will give that a try tomorrow.  I do not really have my RaspiPi v2 setup for development so it will take some time.   I am unfamilar with "go".

rhoppel, 2018-03-14:
I was able to compile.  Here is the tested result:  [https://drive.google.com/open?id=1i749RP_pHzKgoyZvkvQojUszH7HGi5DL](url)
I had to install the following in order to compile:
- libopencv-dev
- libjpeg-dev

works great!

closed 2018-03-14 by rhoppel

gen2brain, 2018-03-15:
Great! You should be able to boost performance a little with `libjpegturbo-dev` instead of plain libjpeg. Not sure if that is correct name of package though. Decoding performance of jpegturbo are much better, but It is also slightly faster with encoding (what cam2ip needs and use).

rhoppel, 2018-03-15:
`libjpegturbo-dev` was not available in the default depositories.    I tried to find one but I gave up.  That is why I settled for `libjpeg-dev`.   I you know of a quality source I will rebuild it.

gen2brain, 2018-03-15:
It is better then to wait for binaries here, probably for new release. Also, binary you have now requires that you install OpenCV runtime libraries on some other RPi (jpeg is probably already isntalled). I prefer to compile binary with static OpenCV and libjpeg-turbo, so it is more portable, just one file to copy and run.