gen2brain/cam2ip #7: High cpu usage?

rolfep, 2018-05-01:
Hi.  I've used this software for a while and it does a nice job with my cheap logitech C270 webcam, thanks.  However, cpu usage for cam2ip is always at the top.

> PID    USER      PR  NI    VIRT    RES   SHR  S  %CPU %MEM     TIME+ COMMAND        
>               
>  7158   rolf        20   0  824140  41904  13576 S  85.2       0.3       5902:38   cam2ip 

This looks to be spread pretty evenly across the 4 cores, according to htop.  A bit earlier this a.m., CPU was more like 75%, maybe due to lower light levels?  I am viewing the stream in Chrome at fullscreen, typically, and it's been running well for about 4 days, this session.  That's much better than when I was using vlc to view the v4l2 device, which would crash frequently, requiring re-starts/re-plugs of the usb cable.  Kudos!

Following are some bits on the hardware and the command I use to start cam2ip.  I wonder if you have any thoughts?  Thanks.

```
$ inxi -C
CPU:       Topology: Quad Core model: Intel Core i5-6500 type: MCP L2 cache: 6144 KB 
           Speed: 3359 MHz min/max: 800/3600 MHz Core speeds (MHz): 1: 3381 2: 3351 3: 3356 
           4: 3320 
[rolf@z170i ~]$ inxi -b
System:    Host: z170i Kernel: 4.14.30-desktop-3.mga6 x86_64 bits: 64 Desktop: N/A 
           Distro: Mageia 6 mga6 
Machine:   Type: Desktop Mobo: ASUSTeK model: Z170I PRO GAMING v: Rev X.0x serial: N/A 
           UEFI [Legacy]: American Megatrends v: 3607 date: 03/21/2018 
CPU:       Quad Core: Intel Core i5-6500 type: MCP speed: 3454 MHz min/max: 800/3600 MHz 
Graphics:  Card-1: Intel HD Graphics 530 driver: i915 v: kernel 
           Display Server: Mageia X.org 11.0 driver: intel,v4l 
           resolution: 1280x800~60Hz, 1152x864~60Hz 
           OpenGL: renderer: llvmpipe (LLVM 3.9 256 bits) v: 3.3 Mesa 17.3.6 
Network:   Card-1: Intel Ethernet Connection I219-V driver: e1000e 
Drives:    HDD Total Size: 268.30 GiB used: 34.69 GiB (12.9%) 
Info:      Processes: 232 Uptime: 6d 11h 43m Memory: 15.54 GiB used: 3.05 GiB (19.6%) 
           Shell: bash inxi: 2.9.11

/home/rolf/cam2ip-1.3-64bit/cam2ip -height 768 -width 1024
```

gen2brain, 2018-05-01:
Those number do look high. On my laptop with 4 cores 640x480 uses about 25% of cpu, and for 1024x768 43%. If you are using /html try to disable webgl with `-nowebgl` option. Also, you can try to raise delay a little with `-delay`, default is 10ms, but can be raised a little and will still be acceptable.

rolfep, 2018-05-02:
This evening, I booted the machine, added -nowebgl to the above command, and opened the url in Chrome.

htop showed about 87% CPU, maybe due to a lot of reflected evening sunshine in the frame?

So, I removed that option and added -delay 20

Now, htop shows about 74% CPU, seems stable there for 10 minutes or so.

I'll see what the occasional animal moving in frame looks like, maybe adjust -delay, and can try something else, if you want.  It's not too big a deal.  Thanks.

gen2brain, 2018-10-28:
You can now try with `native` build, I added support for native V4L implementation, and in my tests it works with much lower CPU usage compared to OpenCV build. Also, I added support for asm base64 on amd64.

rolfep, 2018-10-28:
[image]
Ok, that's the top entry for what I've been using,

`/home/rolf/cam2ip-1.3-64bit/cam2ip -height 768 -width 1024 -delay 30`

where the delay was added to mitigate the high cpu usage.  The scene is early morning, low light levels, which reduced cpu, IIRC.  I'll try with the native build without delay, to begin with:

`/home/rolf/cam2ip-1.5-64bit-native/cam2ip -height 768 -width 1024`

[image]

This is toward the high end of a small range of %CPU that is always above 22%, at this time.

 As light increases, not yet sunlight, the mean %CPU seems to shift upward about one-half percent.
This is the scene:

[image]

The new program looks to be a very significant improvement for me.
Thank you.

rolfep, 2018-10-29:
[image]
After my previous post, I noticed %CPU increasing pretty rapidly to over 40.  It seemed to me perhaps more than just an increase in light level and I shut down the computers, went for a walk.  When I came back, light is, of course, a bit more, no sunlight as yet.  Cam2ip is auto-started by a script in KDE.   Perhaps a breeze is moving branches a bit more.  The initial output is above and a screenie of the scene is here:

[image]

A few minutes later, something has settled down.  I display the scene full-screen in Chrome, usually.

[image]

Letting it run for a while...

Previous CPU issues with kaffeine on this HTPC machine had me suspecting interactions with other apps, especially kodi.  ATM, I opened firefox for a live flash stream and took a screen shot of top:

[image]

After a while, I started a couple of OTA t.v. recordings in kaffeine, closed Firefox with the cpu-hungry stream, started a different http stream in Chrome, and started another instance of Chrome to display the cam2ip scene.  Along in here, the cam2ip scene stopped refreshing and I killed it.  I re-started cam2ip and now have a nice live display.  Initially and for about 14 minutes until now, the initial %CPU looks stable between high 40s to mid 50s.

[image]

[image]

After shutting down the http stream and that instance of Chrome, it sure seems to me the %CPU of cam2ip is affected by other apps running, by the results of top and htop, at least.

[image]

[image]

I logged out and in, giving a new start for cam2ip with no other desktop apps, except Chrome to display the scene.

[image]

The numbers stayed steady for a little while and light is fading, so I want to try one last look after starting kodi.  This is the trio of apps I most commonly use on this HTPC.

[image]

FWIW.  Whether or not these observations are relevant, much less CPU is being used.  Thanks.

closed 2018-11-17 by gen2brain