gen2brain/cam2ip #8: html websocket is 127.0.0.1 when connecting from remote computer (RPI3)

MetallianFR68, 2018-06-09:
I just installed cam2ip on my RPI B3+ and it works well.
My RPI has static IP 192.168.0.60
MJPEG works perfectly when I connect here : http://192.168.0.60:56000/mjpeg

But HTML doesn't work when connecting to http://192.168.0.60:56000/html
I'm connecting from a remote computer on the network 
But the javascript code wants to open websocket 127.0.0.1
"WebSocket connection to 'ws://127.0.0.1:56000/socket' failed: Error in connection establishment:net::ERR_CONNECTION_REFUSED"

It should open ws://192.168.0.60:56000

Am I doing something wrong ?

gen2brain, 2018-06-11:
If you want to bind to specific ip/port use the `-bind` option, if not given it will default to localhost.

gen2brain, 2018-10-10:
Hmm, I tried to use 0.0.0.0 as default, if -bind-addr is not specified, but it doesn't work on all browsers.

gen2brain, 2018-12-02:
Fixed in https://github.com/gen2brain/cam2ip/commit/4d73010019d6eba33d9ca10fed3a11b5dcde53e8 .

closed 2018-12-02 by gen2brain