gen2brain/cam2ip #11: Cam2IP Source Code Not able to compile on Windows 10

mohit83k, 2018-09-05:
i had followed all the steps mentioned for Source Code Compile 
i had installed following things
1. go1.11.windows-amd64.msi
GOPATH: %USERPROFILE%\go
 
2. libjpeg-turbo-2.0.0-gcc.exe 
PATH:  C:\libjpeg-turbo-gcc\bin

3. mingw-w64-install.exe 
PATH: C:\Program Files\mingw-w64\x86_64-7.1.0-posix-seh-rt_v5-rev2\mingw64\bin

4. opencv-2.4.13.6-vc14.exe
PATH: C:\Program Files\mingw-w64\x86_64-7.1.0-posix-seh-rt_v5-rev2\mingw64\opencv\build\x64\vc14\bin

and while trying to compile im getting the error below attached let me know how can i compile successfully
[image]

gen2brain, 2018-09-08:
You can check https://github.com/gen2brain/cam2ip/blob/master/make.bash#L21 .
It is for cross compile for Windows in Linux, but it is example how to point to your directories with CGO_LDFLAGS/CFLAGS . Check where your installed libraries/includes are located and and that to env.
You don't need libjpeg/libjpeg-turbo, you can use `-tags jpeg` for native image/jpeg, but performance is worse with that.

gen2brain, 2018-10-29:
I added support for native capture on Windows, via VFW and without Cgo. Now you can just add `-tags jpeg` to your go get/build command to get fully native binary, and without the need for OpenCV, it is now optional.

closed 2018-10-29 by gen2brain