gen2brain/cam2ip #12: Provide a Docker image

fallais, 2018-09-08:
Hello again,

I am using **cam2ip** instead of motion because motion is using to much CPU all the time, and I love Go by the way. I am using it inside a Docker container. Could be nice to provide it officially no ?

Below the Dockerfile that I am using right now. It could be improved by using alpine and environment variables for example.

```
FROM armhf/ubuntu
LABEL maintainer="Francois ALLAIS <francois.allais@hotmail.com>"

RUN apt-get update && \
    apt-get install -y wget && \
    rm -rf /var/lib/apt/lists/* && \
    mkdir -p /app && \
    cd /app && \
    wget https://github.com/gen2brain/cam2ip/releases/download/1.3/cam2ip-1.3-RPi3.tar.gz && \
    tar zxvpf cam2ip-1.3-RPi3.tar.gz && \
    cp cam2ip-1.3-RPi3/cam2ip ./ && \
    chmod +x cam2ip && \
    rm cam2ip-1.3-RPi3.tar.gz

WORKDIR /app

EXPOSE     80
CMD        [ "/app/cam2ip", "--bind-addr", ":80", "--delay", "2" ]
```

gen2brain, 2018-09-08:
Hello,

In cam2ip releases all binaries are compiled with static libs, so that one binary should just work on those arches. I guess that is ideal for docker, I did use it, but I somehow don't like, in the sense that I don't see a point at publishing cam2ip at docker hub etc.

But, official Dockerfile is always nice to have, but I would prefer something really minimal, not some image that builds 500M for static binary that just works and it weights only a couple of MB.

gen2brain, 2018-09-08:
Sure, I will accept something minimal, but what about arch, that is for RPi3 only?

fallais, 2018-09-08:
For other Linux, I guess that using `frolvlad/alpine-glibc` instead of `armhf/alpine` is fine.

gen2brain, 2018-11-17:
Ok, I added Dockerfile and image in https://github.com/gen2brain/cam2ip/commit/fb1388a10ac0d760e7e02afa008c364cc6d8e154 . Image is just 2M compressed and about 5M uncompresed.

fallais, 2018-11-17:
That is super nice, I will try to test it.
Be careful, you missed a space near the port mapping directive.
It could be good to provide the one for ARM architecture (for Raspberry).

fallais, 2018-11-17:
Ok perfect, perhaps you should notice it on the README ;)

closed 2018-11-17 by fallais