gen2brain/cam2ip #13: Issue with libc6 and the 1.4 version

fallais, 2018-10-17:
Hello,

I am facing this issue with last version : `/lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.27' not found`
Do you know where it could come from please ?

Thanks

gen2brain, 2018-10-17:
Hello,

All binaries are cross compiled, that is probably because I use newer toolchain (glibc version is 2.27) then what is available in raspbian. I will probably have to rebuild whole toolchain for arm6 (RPi 1 and 2) with lower versions. Similar issue was this one https://github.com/gen2brain/cam2ip/issues/6 .

gen2brain, 2018-10-19:
Ok, I uploaded a new binary compiled with toolchain with glibc-2.24, that should work. Can you try now?

fallais, 2018-10-21:
Hello,

Still the same..

```
root@raspberrypi:/app/docker# docker logs docker_cam2ip_1
/app/cam2ip: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
/app/cam2ip: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.27' not found (required by /app/cam2ip)
```

gen2brain, 2018-10-23:
Hmm, cam2ip-1.4-RPi.tar.gz binary should not have such symbols anymore. I can try to build everything once more, if you are willing to test it.

fallais, 2018-10-23:
Yes I can test

gen2brain, 2018-10-28:
Can you try again with https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-RPi-native.tar.gz , I added suport for V4L native implementation on Linux/RPi. On desktop it works with lower CPU usage compared to version with OpenCV.

fallais, 2018-10-28:
Hello,

I have this issue now :

```
root@raspberrypi:/app/docker# docker logs docker_cam2ip_1
/app/cam2ip: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
/app/cam2ip: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
/app/cam2ip: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
/app/cam2ip: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
/app/cam2ip: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
/app/cam2ip: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
/app/cam2ip: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
```

gen2brain, 2018-10-28:
Ok, new build is uploaded, just download again, it should be ok now.

fallais, 2018-10-28:
I tried again, I face the same issue

gen2brain, 2018-10-28:
Are you sure that is not related to some cache? The last binary I uploaded is compiled with static libjpeg.a, and symbols are in binary:
```
cam2ip-1.5-RPi-native # armv6j-hardfloat-linux-gnueabi-nm -D cam2ip | grep " jpeg"
         U jpeg_CreateCompress
         U jpeg_CreateDecompress
         U jpeg_destroy_compress
         U jpeg_destroy_decompress
         U jpeg_finish_compress
         U jpeg_finish_decompress
         U jpeg_mem_dest
         U jpeg_mem_src
         U jpeg_read_header
         U jpeg_read_scanlines
         U jpeg_set_defaults
         U jpeg_set_quality
         U jpeg_start_compress
         U jpeg_start_decompress
         U jpeg_std_error
         U jpeg_write_scanlines
```

Anyway, you can try also with https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-RPi-native-jpeg.tar.gz I just added, it is a completely static binary compiled without CGo support.

fallais, 2018-10-28:
I tried again, still the same. I tried the other one and it is working, but the default quality of the image is very poor.

Note : the rotate works :-)

fallais, 2018-10-29:
Is it normal that the more I increase the width and height, the more it is zoomed ? It did not work like this before, it was just increasing the quality, which is what I want

fallais, 2018-11-16:
Up :-)

gen2brain, 2018-11-17:
Can you grab a screenshot? I just tried in Linux OpenCV build and native build, both with 1024x768 and I don't see a difference.

fallais, 2018-11-17:
I tried but when trying to rebuild the Dockerfile, here is my issue :

```
--2018-11-17 11:55:38--  https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-RPi-native-jpeg.tar.gz
Resolving github.com (github.com)... 140.82.118.4, 140.82.118.3
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-11-17 11:55:43 ERROR 404: Not Found.
```

Did you remove the binary ?

gen2brain, 2018-11-17:
That is a https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-RPi-nocgo.tar.gz now.

fallais, 2018-11-17:
With this Dockerfile.

```
FROM armhf/ubuntu
LABEL maintainer="xxxx"

RUN apt-get update && \
    apt-get install -y wget && \
    rm -rf /var/lib/apt/lists/* && \
    mkdir -p /app && \
    cd /app && \
    wget https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-RPi-native-jpeg.tar.gz && \
    tar zxvpf cam2ip-1.5-RPi-native-jpeg.tar.gz && \
    cp cam2ip-1.5-RPi-native-jpeg/cam2ip ./ && \
    chown nobody:nogroup cam2ip && \
    chmod +x cam2ip && \
    rm cam2ip-1.5-RPi-native-jpeg.tar.gz

WORKDIR /app

EXPOSE     80
CMD        [ "/app/cam2ip", "--bind-addr", ":80", "--width", "1280", "--height", "720", "--delay", "2000", "--rotate", "180" ]
```

[image]

With this Dockerfile :

```
FROM armhf/ubuntu
LABEL maintainer="xxxx"

RUN apt-get update && \
    apt-get install -y wget && \
    rm -rf /var/lib/apt/lists/* && \
    mkdir -p /app && \
    cd /app && \
    wget https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-RPi-nocgo.tar.gz && \
    tar zxvpf cam2ip-1.5-RPi-nocgo.tar.gz && \
    cp cam2ip-1.5-RPi-nocgo/cam2ip ./ && \
    chown nobody:nogroup cam2ip && \
    chmod +x cam2ip && \
    rm cam2ip-1.5-RPi-nocgo.tar.gz

WORKDIR /app

EXPOSE     80
CMD        [ "/app/cam2ip", "--bind-addr", ":80", "--width", "720", "--height", "480", "--delay", "2000", "--rotate", "180" ]
```

[image]

PS : It is a chicken coop ^^