gen2brain/cam2ip #15: The "native VFW implementation on Windows" binary doesn't seem to work

fireattack, 2019-04-24:
I tried both "cam2ip-1.5-64bit.zip" and "cam2ip-1.5-32bit.zip" version on 4 different computers (2 Win 7 & 2 Win10, all 64-bit) with USB cameras from various vendors, none of them work.

The symptom is: the program started fine, and the camera would be activated for a second and then it immediately went off. 

There is no apparent error on the console, until you open http://localhost:56000/mjpeg, when it will start to throw `2019/04/24 14:34:28 mjpeg: read: camera: can not retrieve frame: EOF` errors constantly. 

The "cam2ip-1.5-64bit-cv2.zip" version works flawlessly.

It is not a big deal as I made it work, but think developer may be want to know this. 

Thanks!

fireattack, 2019-04-24:
Small note: this is under my impression that users don't need to install anything additionally for " Video for Windows (VfW) framework" to run the binary - I didn't find anything to download for that, at least. 

If this is not the case please correct me.

gen2brain, 2019-04-30:
I only tested `Video for Windows` in VMs, both on Win7 and Win10. It is a very old API but is available via syscalls. Newer versions of Windows have some dll wrappers that wrap new technologies and APIs and make them available for `VfW` but I don't know all the details. I remember there was some tool from such wrappers to check and maybe enable? support, really not sure.

Xart3mis, 2022-11-02:
same issue here

ricolk, 2023-01-18:
Also using cam2ip-1.6-64bit-cv, CV version on Win10, I can get only a frame with a http://localhost:56000/jpeg request.
After every further request I get this error message.

`jpeg: read: camera: can not retrieve frame`

[image]

Restarting the binary, it works for a single frame only.

exxxt4zy, 2023-10-08:
use cv2 version