gen2brain/cam2ip #20: jpeg: read: camera: invalid JPEG format: uninitialized Huffman table

Fmstrat, 2019-10-05:
Using a Logitech webcam I'm getting:
```
2019/10/05 14:23:49 jpeg: read: camera: invalid JPEG format: uninitialized Huffman table
```
I'm guessing this is a similar problem to this fix:

https://github.com/blackjack/webcam/issues/7#issuecomment-231543221

gen2brain, 2019-10-05:
It looks like it is valid for MJPEG frames to omit the Huffman table and expect some default to be used. Browsers do that, they will load such an image even if not valid, probably by adding some default table.

I am guessing you are on Linux and don't use provided binaries but you compiled with native image/jpeg support and with native V4L, right? It will probably work with OpenCV build, with libjpegturbo it will probably be a different error.

Problem is, it is not correct to just add tables as in your link, will need to detect first if that is the issue, my cam, for example, returns valid JPEG with Huffman tables from MJPEG. One way is to match if the error string contains `Huffman` I guess, but that is ugly. Something like this would be great https://github.com/golang/go/issues/10447 but it is not implemented.

Fmstrat, 2019-10-05:
Ubuntu 18.04, using the latest Docker HUB release. Camera works as expected in `cheese` and with `motion`. Hope that helps!

gen2brain, 2019-10-05:
Yes, docker image is built without CGO, and uses native image/jpeg and native V4L.

Can you try with binaries from releases, and paste error messages, this one https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-64bit.tar.gz uses libjpegturbo and V4L, and this one https://github.com/gen2brain/cam2ip/releases/download/1.5/cam2ip-1.5-64bit-cv2.tar.gz libjpegturbo and OpenCV.

carlos-sarmiento, 2019-12-13:
Is there any reason not to include the solution to this issue in the docker image release?

gen2brain, 2019-12-16:
@carlos-sarmiento the solution should not be included unconditionally, it should probably check error string for Huffman table, a little ugly, but if nothing else possible I agree. I don't have the webcam that behaves like that or time to work on this now. PRs are welcome.

davralin, 2020-12-30:
I have a RPI3 with a Logitech C310, and docker-image gave me `uninitialized Huffman table`.

I used cam2ip-1.6-RPi3.tar.gz and console is spammed with the following during a stream:
```
Corrupt JPEG data: 1 extraneous bytes before marker 0xd6
Corrupt JPEG data: 2 extraneous bytes before marker 0xd6
Corrupt JPEG data: 2 extraneous bytes before marker 0xd2
Corrupt JPEG data: 1 extraneous bytes before marker 0xd4
Corrupt JPEG data: 1 extraneous bytes before marker 0xd4
Corrupt JPEG data: 1 extraneous bytes before marker 0xd3
Corrupt JPEG data: 3 extraneous bytes before marker 0xd3
Corrupt JPEG data: 1 extraneous bytes before marker 0xd2
Corrupt JPEG data: 3 extraneous bytes before marker 0xd7
Corrupt JPEG data: 1 extraneous bytes before marker 0xd4
Corrupt JPEG data: 1 extraneous bytes before marker 0xd7
Corrupt JPEG data: 2 extraneous bytes before marker 0xd0
Corrupt JPEG data: 1 extraneous bytes before marker 0xd3
Corrupt JPEG data: 1 extraneous bytes before marker 0xd0
Corrupt JPEG data: 1 extraneous bytes before marker 0xd0
Corrupt JPEG data: 1 extraneous bytes before marker 0xd0
Corrupt JPEG data: 1 extraneous bytes before marker 0xd1
Corrupt JPEG data: 2 extraneous bytes before marker 0xd6
Corrupt JPEG data: 1 extraneous bytes before marker 0xd4
```

I can't really help any more than that, but errormessages aside, the camera does stream.

davralin, 2021-01-14:
Same issue on amd64 with a Logitech C510.

Doesn't work with Docker, works just fine with the relevant release - I created davralin/cam2ip witch contains the amd64-build for now, as I have to use this in a container (deployed on k8s).

gen2brain, 2021-01-14:
@davralin so you use OpenCV or native build?

davralin, 2021-01-15:
> @davralin so you use OpenCV or native build?

Native build I'd guess?

Source is here: https://github.com/davralin/cam2ip-docker/blob/main/run.sh

T0MuX, 2021-03-06:
Hi, same issue with https://github.com/gen2brain/cam2ip/releases/download/1.6/cam2ip-1.6-RPi.tar.gz

When I access to http://localhost:56000/html, I got this :
`2021/03/06 22:48:20 socket: read: camera: invalid JPEG format: uninitialized Huffman table`

When I access to http://localhost:56000/jpeg, I got this :
`2021/03/06 22:50:23 jpeg: read: camera: invalid JPEG format: uninitialized Huffman table`

And when I access to http://localhost:56000/mjpeg, I got this big one :
```
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:28 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:29 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:29 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:29 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:29 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:29 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
2021/03/06 22:52:29 mjpeg: read: camera: invalid JPEG format: uninitialized Huffman table
...
```

T0MuX, 2021-03-06:
Okay, the "edit button" of Github doesn't work for me so I post again...

I forgot to specify : I'm using ArchlinuxARM on a Raspberry 1B. Also, I miswrote : I was using cam2ip-1.6-RPi-nocgo.tar.gz
But since, I "solved" this switching to cam2ip-1.6-RPi.tar.gz :)

5vr, 2024-02-16:
> Something like this would be great [golang/go#10447](https://github.com/golang/go/issues/10447) but it is not implemented.

https://github.com/golang/go/issues/10447#issuecomment-1506257410

How to use it now?

gen2brain, 2024-02-16:
To use what, something that is not implemented? I guess you would have to implement it first. Look for alternative approach above and send PR if you can.

gen2brain, 2025-06-15:
I added support for `jpegli`, which uses an option that disables Huffman tables and uses arithmetic coding instead. To build it, add `-tags jpegli` to the build command. This should probably fix the issue, although it will work a little slower, but it is still acceptable.