gen2brain/cam2ip #21: Only open camera when capturing

orulz, 2020-05-15:
This is a fantastic tool! It would work better for my case if it if it did not grab the camera as soon as the server is started, but instead only when it is actively serving video or images. This might result in some lag time when starting up, but this would not be a problem for me. Leaving the camera off when the server is not in use would be helpful.

Not familiar with go or opencv, but it looks that the camera is initialized in New() https://github.com/gen2brain/cam2ip/blob/19ea541157a10d42eb53f2aeaf75cdc63fac2cb8/camera/camera_cv2.go#L26

Maybe that initialization could be (optionally) deferred to Read() ??

gen2brain, 2020-05-15:
It seems like a nice addition. This should be done by handlers, not Read(), and if option enabled main should not initialize the camera. I think it should not be a big job but need to check first, not sure when. What do you think option should be called, `lazy` sounds ok?

orulz, 2020-05-15:
"Deferred camera initialization" is the most descriptive way, but that doesn't fit well into a short CLI option, so "lazy" works. Kind of fun.

orulz, 2020-05-15:
One thing to think about is, what happens when multiple clients are accessing the camera at once.

Would need to maintain a client counter and only release the camera when the client counter reaches zero.

gen2brain, 2020-05-15:
Every handler will anyway check if camera is nil/null on every request and will initialize it if it is not, so one client closes tab/connection and it closes the camera, the other client on the next request will again initialize the camera. Needs to be tested but I would not want to complicate this much.

orulz, 2020-05-15:
If somebody is viewing MJPEG stream they would not want to have the camera turned off underneath them by somebody capturing a single JPEG... This would result in a glitch in the video stream as the camera has to be reinitialized for the next MJPEG frame. I guess that's not the end of the world, but not perfect. Or is that not an issue for some other reason I'm not seeing? Oh well! It's your project, your call!

Thanks again for this project and thanks for being so responsive!

orulz, 2020-05-15:
Maybe initialize as you suggest, and handle closing the camera by a ConnState hook ( https://golang.org/pkg/net/http/#ConnState ) that increments a counter on StateNew, decrements on StateClosed (maybe StateHijacked too) and closes the camera whenever the counter transitions to zero.

Again not a Go expert, just reading the docs...