gen2brain/cam2ip #24: android build

sam-cts, 2021-01-19:
Hi there, 

I am interested to build this application for android device, I saw you do have android support, however I am having trouble to build it. Can you please kindly share the build steps for the android build?

Thanks.

gen2brain, 2021-01-19:
Hi, you have an example here https://github.com/gen2brain/cam2ip/blob/master/make.bash#L106, you need to install the Android toolchain first. You also must have rooted devices so you can start it from the shell.
It is still a work in progress, it uses `camera2` API but I don't get a callback on my current phone so I can not test it. Btw. there is a was also to build opencv2 version and use it for the native camera on Android, but up to version 4.0 I think, so only for very old devices.

sam-cts, 2021-01-20:
@gen2brain thank you very much, I got it to build and running on rooted device now. However it is not able to retrieve frame from camera. does it require some camera permission setting?

gen2brain, 2021-01-20:
Check what your camera supports, there are apps like this one https://play.google.com/store/apps/details?id=com.airbeat.device.inspector, and as I said, it is still WIP and not tested.