gen2brain/cam2ip #29: docker setup

st3ffl0n, 2021-03-02:
Hello

I have an error during run docker :

docker: Error response from daemon: error gathering device information while adding custom device "/dev/video0": no such file or directory.
ERRO[0000] error waiting for container: context canceled 

I also tried finding the right identifier for my device (dev/ttys001) but this also lead to the same error message
Also running the OpenCV binary works and the steam is accessible at http://localhost:56000/html
  
Do you have any suggestions on what I could try?

gen2brain, 2021-03-04:
Hello,

The message is clear, there is no /dev/video0, what are your exact steps and device name? 
Btw. if static binary works and is just one file, not sure why insist on docker? In this case image is built from `scratch` so it is again just one binary, but usually you get the whole distro together with binary.

st3ffl0n, 2021-03-04:
I want to use docker to make the stream available on my local network and also set a password

I unplugged my webcam, then:
ls -lha /dev/* > unplugged.txt 
plugged the webcam back in, then:
ls -lha /dev/* > plugged.txt
compare the devices to find the right name
vimdiff plugged.txt unplugged.txt

my device name is /dev/ttys001, but trying to run docker with this doesn't work
then I tried everything like device=/dev/ttys001:/dev/ttys001 ... device=/dev/video0:/dev/ttys001 ...
I alway get the same error message

gen2brain, 2021-03-04:
Did you try with `--privileged` or `--network host`, maybe `--device=/dev:/dev`? Is that a USB or serial camera?

st3ffl0n, 2021-03-04:
its a cheap usb camera https://www.amazon.de/gp/product/B08M99YWG9/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1

with `--privileged` it says "no camera at index 0" `docker run --privileged --device=/dev/ttys001:/dev/ttys001 -p56000:56000 -it gen2brain/cam2ip`

the other options don't work

st3ffl0n, 2021-09-07:
I got it now working on a Raspberry pi. The only issue is that he doesn't recognise my .htpasswd path

I always get "stat gen2brain/cam2ip/htpasswd: no such file or directory"
I tried many path but don't seem to be able to find the base directory from which docker is looking

home/pi/Downloads/auth.htpasswd doesn't work.. 

Where do I have to put the file?

closed 2022-02-03 by st3ffl0n