gen2brain/cam2ip #31: Random crash on Mac OS after running few minutes

nvdhau-v2cloud, 2021-04-22:
I tried the precompiled for Mac OS. It works but randomly crashes after a few minutes.

Command: `cam2ip' -bind-addr :62001 -delay 100`
OS: Mac OS 11.2.3

Logs:

> Listening on :62001
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x7fff274a81c9]
runtime stack:
runtime.throw(0x49a96c4, 0x2a)
	/home/milann/go/src/runtime/panic.go:774 +0x72
runtime.sigpanic()
	/home/milann/go/src/runtime/signal_unix.go:378 +0x47c
goroutine 40 [syscall]:
runtime.cgocall(0x43b7fe0, 0xc00025f7a0, 0x1)
	/home/milann/go/src/runtime/cgocall.go:128 +0x5b fp=0xc00025f770 sp=0xc00025f738 pc=0x40058eb
github.com/gen2brain/go-opencv/opencv._Cfunc_cvGrabFrame(0x52056c0, 0x0)
	_cgo_gotypes.go:2108 +0x49 fp=0xc00025f7a0 sp=0xc00025f770 pc=0x4200649
github.com/gen2brain/go-opencv/opencv.(*Capture).GrabFrame.func1(0x52056c0, 0xc000092e00)
	/home/milann/golang/pkg/mod/github.com/gen2brain/go-opencv@v0.0.0-20191005190506-bf186fc94f7a/opencv/highgui.go:497 +0x56 fp=0xc00025f7d8 sp=0xc00025f7a0 pc=0x4207c06
github.com/gen2brain/go-opencv/opencv.(*Capture).GrabFrame(0x52056c0, 0xc000146030)
	/home/milann/golang/pkg/mod/github.com/gen2brain/go-opencv@v0.0.0-20191005190506-bf186fc94f7a/opencv/highgui.go:497 +0x2b fp=0xc00025f7f8 sp=0xc00025f7d8 pc=0x42056ab
github.com/gen2brain/cam2ip/camera.(*Camera).Read(0xc000086580, 0xc00025f960, 0x4a2f480, 0xc0000a2140, 0x0)
	/home/milann/Projects/cam2ip/camera/camera_cv2.go:43 +0x44 fp=0xc00025f8a8 sp=0xc00025f7f8 pc=0x42099a4
github.com/gen2brain/cam2ip/handlers.(*MJPEG).ServeHTTP(0xc0000a2520, 0x4a385a0, 0xc0003e4000, 0xc0003d2000)
	/home/milann/Projects/cam2ip/handlers/mjpeg.go:59 +0x5cf fp=0xc00025faf0 sp=0xc00025f8a8 pc=0x43b0e2f
github.com/gen2brain/cam2ip/server.newAuthHandler.func1(0x4a385a0, 0xc0003e4000, 0xc0003d2000)
	/home/milann/Projects/cam2ip/server/server.go:85 +0x71 fp=0xc00025fb90 sp=0xc00025faf0 pc=0x43b1f51
net/http.HandlerFunc.ServeHTTP(0xc0000a2540, 0x4a385a0, 0xc0003e4000, 0xc0003d2000)
	/home/milann/go/src/net/http/server.go:2007 +0x44 fp=0xc00025fbb8 sp=0xc00025fb90 pc=0x4369b64
net/http.(*ServeMux).ServeHTTP(0x4d9d6a0, 0x4a385a0, 0xc0003e4000, 0xc0003d2000)
	/home/milann/go/src/net/http/server.go:2387 +0x1bd fp=0xc00025fc18 sp=0xc00025fbb8 pc=0x436ba3d
net/http.serverHandler.ServeHTTP(0xc000118000, 0x4a385a0, 0xc0003e4000, 0xc0003d2000)
	/home/milann/go/src/net/http/server.go:2802 +0xa4 fp=0xc00025fc48 sp=0xc00025fc18 pc=0x436cfb4
net/http.(*conn).serve(0xc0003b8000, 0x4a39120, 0xc0003ca000)
	/home/milann/go/src/net/http/server.go:1890 +0x875 fp=0xc00025ffc8 sp=0xc00025fc48 pc=0x4368955
runtime.goexit()
	/home/milann/go/src/runtime/asm_amd64.s:1357 +0x1 fp=0xc00025ffd0 sp=0xc00025ffc8 pc=0x405c781
created by net/http.(*Server).Serve
	/home/milann/go/src/net/http/server.go:2927 +0x38e
goroutine 1 [IO wait]:
internal/poll.runtime_pollWait(0x50b6f68, 0x72, 0x0)
	/home/milann/go/src/runtime/netpoll.go:184 +0x55
internal/poll.(*pollDesc).wait(0xc0000f8098, 0x72, 0x0, 0x0, 0x4999adc)
	/home/milann/go/src/internal/poll/fd_poll_runtime.go:87 +0x45
internal/poll.(*pollDesc).waitRead(...)
	/home/milann/go/src/internal/poll/fd_poll_runtime.go:92
internal/poll.(*FD).Accept(0xc0000f8080, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/home/milann/go/src/internal/poll/fd_unix.go:384 +0x1f8
net.(*netFD).accept(0xc0000f8080, 0xc0c882361d, 0x6d6cc882361d, 0x100000001)
	/home/milann/go/src/net/fd_unix.go:238 +0x42
net.(*TCPListener).accept(0xc0000a25e0, 0x608200f4, 0xc000050bd8, 0x40b5c56)
	/home/milann/go/src/net/tcpsock_posix.go:139 +0x32
net.(*TCPListener).Accept(0xc0000a25e0, 0xc000050c28, 0x18, 0xc000000180, 0x436d47e)
	/home/milann/go/src/net/tcpsock.go:261 +0x47
net/http.(*Server).Serve(0xc000118000, 0x4a382e0, 0xc0000a25e0, 0x0, 0x0)
	/home/milann/go/src/net/http/server.go:2896 +0x286
github.com/gen2brain/cam2ip/server.(*Server).ListenAndServe(0xc0000b4120, 0xc0000aa008, 0x499c9b4)
	/home/milann/Projects/cam2ip/server/server.go:71 +0x3e8
main.main()
	/home/milann/Projects/cam2ip/cmd/cam2ip/main_cv.go:80 +0x66e
goroutine 52 [IO wait, 2 minutes]:
internal/poll.runtime_pollWait(0x50b6e98, 0x72, 0xffffffffffffffff)
	/home/milann/go/src/runtime/netpoll.go:184 +0x55
internal/poll.(*pollDesc).wait(0xc0003d6018, 0x72, 0x0, 0x1, 0xffffffffffffffff)
	/home/milann/go/src/internal/poll/fd_poll_runtime.go:87 +0x45
internal/poll.(*pollDesc).waitRead(...)
	/home/milann/go/src/internal/poll/fd_poll_runtime.go:92
internal/poll.(*FD).Read(0xc0003d6000, 0xc0003c8041, 0x1, 0x1, 0x0, 0x0, 0x0)
	/home/milann/go/src/internal/poll/fd_unix.go:169 +0x22b
net.(*netFD).Read(0xc0003d6000, 0xc0003c8041, 0x1, 0x1, 0x0, 0x0, 0x0)
	/home/milann/go/src/net/fd_unix.go:202 +0x4f
net.(*conn).Read(0xc0000aa088, 0xc0003c8041, 0x1, 0x1, 0x0, 0x0, 0x0)
	/home/milann/go/src/net/net.go:184 +0x68
net/http.(*connReader).backgroundRead(0xc0003c8030)
	/home/milann/go/src/net/http/server.go:677 +0x58
created by net/http.(*connReader).startBackgroundRead
	/home/milann/go/src/net/http/server.go:673 +0xd4

Do you have any ideas about the issue? 

Thanks!

gen2brain, 2021-04-25:
Some backtrace would maybe be useful, precompiled binary is built with OpenCV 2.x. Perhaps you can build with 4.x version on your machine. Also, macOS binary is cross-compiled on Linux, but that should not be an issue.

nvdhau-v2cloud, 2021-04-26:
How can I get some backtrace for you?

gen2brain, 2021-04-27:
Just install gdb, and `gdb ./cam2ip`, then `run`, and after it crashes `bt`. But, it looks like it crashes in OpenCV cgo call, so I doubt I can do anything. I can do both OpenCV 2.x and 4.x builds when I do next release, but I don't have much time to do that right now.

nvdhau-v2cloud, 2021-04-30:
I cannot run gdb, below is the gdb log
```
$ gdb ./cam2ip
GNU gdb (GDB) 10.2
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-apple-darwin20.3.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./cam2ip...

warning: Could not open OSO archive file "/usr/x86_64-apple-darwin14/SDK/MacOSX10.10.sdk/usr/lib/libpng.a"
(No debugging symbols found in ./cam2ip)
(gdb) 
```

I tried to compile cam2ip with gocv (OpenCV 4.x) on MacOS by command `CGO_ENABLED=1 GOOS=darwin GOARCH=amd64 go build -tags cv4 -o cam2ip-ldflags "-linkmode external -s -w" github.com/gen2brain/cam2ip/cmd/cam2ip` with different configs:

1. go 1.13 (gocv 0.20.0)
2. go 1.16 (gocv 0.20.0)
3. go 1.13 and updated all packages in go.mod (gocv 0.27.0)
4. go 1.16 and updated all packages in go.mod (gocv 0.27.0)

I tested each cam2ip binary with a built-in webcam and a USB webcam Logitech C525 for about 20 minutes each webcam. I found that all of them crashes but with different frequencies; 1 and 2 crashed more frequently than 3 and 4. The binary number 3 is the most stable. In addition, the built-in webcam is more stable than the external USB webcam. Below are the logs of them

1. [cam2ip-go1.13-opencv4-failed.log](https://github.com/gen2brain/cam2ip/files/6402280/cam2ip-go1.13-opencv4-failed.log)
2. [cam2ip-1.16-opencv4-failed.log](https://github.com/gen2brain/cam2ip/files/6402279/cam2ip-1.16-opencv4-failed.log)
3. [cam2ip-go1.13-updated-module.log](https://github.com/gen2brain/cam2ip/files/6402283/cam2ip-go1.13-updated-module.log)
4. [cam2ip-go1.16-updated-modules-2.log](https://github.com/gen2brain/cam2ip/files/6402284/cam2ip-go1.16-updated-modules-2.log)

However, when I copied those cam2ip binaries to other Mac, none of them can run because they were compiled with dynamic shared libraries.

```
MacBook-Pro:cam2ip brendato$ ./go1.13
go1.13/         go1.13-updated/ 
MacBook-Pro:cam2ip brendato$ ./go1.13
go1.13/         go1.13-updated/ 
MacBook-Pro:cam2ip brendato$ ./go1.13-updated/cam2ip
dyld: Library not loaded: /usr/local/opt/opencv/lib/libopencv_gapi.4.5.dylib
  Referenced from: /Users/brendato/Downloads/cam2ip/./go1.13-updated/cam2ip
  Reason: image not found
Abort trap: 6
```

I tried altering the build command to make a static build but still cannot make it. Can you guide me to compile the static cam2ip?

gen2brain, 2021-04-30:
Try to just type `run` in gdb console, then after it crashes `bt`. Anyway, seems strange there are so many crashes, people run this for months without any issues, e.g. on RPi. The only difference is that one can use the native build on Linux, while on macOS it must be cgo build.