gen2brain/cam2ip #38: Goost lib?

dpalmasan, 2022-10-15:
I get an import error related to `base64` from goost namespace. It seems goost.org does not work. Wouldn't it be better using the standard base64? I have a fork with the fix if interested.

```
go get goost.org/encoding/base64 
go get: unrecognized import path "goost.org/encoding/base64": https fetch: Get "https://goost.org/encoding/base64?go-get=1": dial tcp: lookup goost.org on 172.25.192.1:53: no such host
```
If not, is there a workaround for this?

gen2brain, 2022-10-15:
Fixed in https://github.com/gen2brain/cam2ip/commit/f58b475549c88d66539e1ba1300fbc082bc8c1c6.

closed 2022-10-15 by gen2brain

gen2brain, 2022-10-15:
goost.org implements base64 in assembly, I prefer to use it where is available (on amd64). For some time it was available on github.com/libgoost, but now that is also gone, it now uses the fork I had locally.