gen2brain/cam2ip #39: Add ability to choose camera by ID

bronzegoddess, 2022-10-23:
Currently when I run the program I go through a series of indexes until I find the right camera on my mac. 
`-index 1, 2, 3 etc
`
It would be nice if we could call the camera by its Unique ID. 
Example:  `Unique ID:	0x4015000005ac1112`

Also, it would be even better if the program would tell us what those Unique Ids are (on our system)

**For Example**

1. Start the program by running it via index
`/Users/bronzemedia/Downloads/cam2ip -index 2  -bind-addr ":4444" -width 1920 -height 1080`
2. The system responds with:
```
Listening on. :4444
CameraID.     : 0x4015000005ac1112  
```
3. Then next time we can call it like this:
`/Users/bronzemedia/Downloads/cam2ip -cameraid  "0x4015000005ac1112"  -bind-addr ":4444" -width 1920 -height 1080`

gen2brain, 2022-10-26:
Unfortunately, OpenCV which is used for macOS doesn't provide a name or ID, only an index. Even in native Linux and Windows implementations, I don't see a way, perhaps it is possible to dig that info on Windows. At least, on Linux is possible to always map the camera indexes via UDEV so they don't change.

bronzegoddess, 2022-11-02:
Thank you for the clarification. 
I will look into disabling other cameras on my Mac so that the webcam is the only option.

closed 2025-06-15 by gen2brain