gen2brain/cam2ip #40: jpeglib.h: No such file or directory

HenryFBP, 2022-10-30:
Hello,

I get this error when installing this Go package.

```
root@crustbook:/home/henry/.motion#     go get -v github.com/gen2brain/cam2ip/cmd/cam2ip
github.com/antonini/golibjpegturbo
github.com/klauspost/compress/flate
# github.com/antonini/golibjpegturbo
/root/go/src/github.com/antonini/golibjpegturbo/decode.go:32:10: fatal error: jpeglib.h: No such file or directory
   32 | #include <jpeglib.h>
      |          ^~~~~~~~~~~
compilation terminated.
# github.com/klauspost/compress/flate
/root/go/src/github.com/klauspost/compress/flate/deflate.go:135:23: cannot convert d.window (type []byte) to type *[32768]byte
/root/go/src/github.com/klauspost/compress/flate/deflate.go:135:56: cannot convert d.window[windowSize:] (type []byte) to type *[32768]byte
/root/go/src/github.com/klauspost/compress/flate/fast_encoder.go:93:28: cannot convert e.hist (type []byte) to type *[32768]byte
/root/go/src/github.com/klauspost/compress/flate/fast_encoder.go:93:63: cannot convert e.hist[offset:] (type []byte) to type *[32768]byte
/root/go/src/github.com/klauspost/compress/flate/huffman_bit_writer.go:794:17: cannot convert w.literalFreq[:] (type []uint16) to type *[256]uint16
/root/go/src/github.com/klauspost/compress/flate/huffman_bit_writer.go:796:16: cannot convert w.literalFreq[256:] (type []uint16) to type *[32]uint16

```

Os info:

```
/root/go/src/github.com/klauspost/compress/flate/fast_encoder.go:93:63: caroot@crustbook:/home/henry/.motion# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@crustbook:/home/henry/.motion# 
```

...

```
root@crustbook:/home/henry/.motion# apt search jpegli*
Sorting... Done
Full Text Search... Done
```

gen2brain, 2022-10-30:
You need to install the requirements, i.e. libjpeg-turbo headers aka dev/devel package, or just use `-tags jpeg` to use native `image/jpeg`.

HenryFBP, 2022-10-30:
@gen2brain Thanks!

closed 2025-06-15 by gen2brain