gen2brain/cam2ip #42: Can't get higher than 5FPS

torindkflt, 2022-12-09:
Greetings. I have a Windows 10 system with two 1080p webcams connected to it running cam2ip for both cameras. The cameras support up to 30FPS at 1080p, but cam2ip will not run them at any higher than 5FPS no matter what value I use for the -delay option. I don't need to run them at the full 30FPS, but I'd like to be able to run them at at least 10FPS. Is there an additional setting I'm overlooking?

Thank you.

torindkflt, 2022-12-09:
I'm using the 64-bit OpenCV version, as the VfW version indeed does not seem to work with my cameras. Disabling WebGL had no effect. I'm streaming the MJPEG streams to Agent DVR (replacement for iSpy security), and have verified that Agent is configured to read the stream at 10FPS, so the bottleneck seems to be with cam2ip. The computer it's running on is holding at about 40% CPU usage and 60% RAM usage, so it's seemingly not a CPU or RAM limitation.

gen2brain, 2022-12-09:
Looks like you can set the delay to 0 then, and you should probably increase that Agent FPS.

torindkflt, 2022-12-09:
No luck. I set the delay to 0 and configured Agent to record at 30FPS (although eventually I only need 10-15FPS), and the cameras are still only being streamed at 5FPS. I know the cameras do support 30FPS because if I open them directly on that computer using ffplay, it displays them at 30FPS just fine.

Network bandwidth isn't an issue. Both this computer and the one running Agent are connected to Gig-E, and Task Manager shows the system running cam2ip only streaming at 30mbit/sec, so plenty of bandwidth left over.

gen2brain, 2022-12-09:
What is the FPS if you try it on localhost without Agent, i.e. `ffplay -i http://localhost:56000/mjpeg`?

torindkflt, 2022-12-09:
Using ffplay through localhost still returns about 5FPS. If I close cam2ip and connect ffplay directly to the camera (`ffplay -f dshow -video_device_number 0 -i video="@device_pnp_\\?\usb#vid_32e4&pid_9230&mi_00#6&2ba1ddea&1&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global"`), it plays at full 30FPS.

fluoroom, 2023-01-05:
I'm facing this issue too using Windows x64 build