gen2brain/cam2ip #45: Cannot install using go get

MaddyPhi, 2023-04-11:
When I try to install using `go get -v -tags cv2 github.com/gen2brain/cam2ip/cmd/cam2ip` I get the following:
```
# pkg-config --cflags  -- opencv
Package opencv was not found in the pkg-config search path.
Perhaps you should add the directory containing 'opencv.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opencv' found
pkg-config: exit status 1
# github.com/klauspost/compress/flate
go/src/github.com/klauspost/compress/flate/deflate.go:135:23: cannot convert d.window (type []byte) to type *[32768]byte
go/src/github.com/klauspost/compress/flate/deflate.go:135:56: cannot convert d.window[windowSize:] (type []byte) to type *[32768]byte
go/src/github.com/klauspost/compress/flate/fast_encoder.go:93:28: cannot convert e.hist (type []byte) to type *[32768]byte
go/src/github.com/klauspost/compress/flate/fast_encoder.go:93:63: cannot convert e.hist[offset:] (type []byte) to type *[32768]byte
go/src/github.com/klauspost/compress/flate/huffman_bit_writer.go:794:17: cannot convert w.literalFreq[:] (type []uint16) to type *[256]uint16
go/src/github.com/klauspost/compress/flate/huffman_bit_writer.go:796:16: cannot convert w.literalFreq[256:] (type []uint16) to type *[32]uint16
```

When I do `go get -v github.com/gen2brain/cam2ip/cmd/cam2ip` I get the same without the opencv issue.

When I run one of the binaries offered, I get `-bash: ./cam2ip: No such file or directory` except for if I use the rpi static version which seems to work. However, when running that version it would seem that the frame rate is very slow.

For some context, this is the system I am working with
```
pi@crafthub:~/cam2ip-1.6-RPi-nocgo $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@crafthub:~/cam2ip-1.6-RPi-nocgo $ v4l2-ctl --list-formats-ext --device /dev/video0
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 2048x1536
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 2048x1536
                        Interval: Discrete 0.200s (5.000 fps)
pi@crafthub:~/cam2ip-1.6-RPi-nocgo $ 
```
Any help you can offer, I would appreciate. Thank you in advance.

mmokrejs, 2024-02-17:
It worked for me:

```
$ export GOPATH=$HOME/apps/cam2ip
go install github.com/gen2brain/cam2ip/cmd/cam2ip@latest
go: downloading github.com/gen2brain/cam2ip v0.0.0-20240130151524-b302c77f20e3
go: downloading github.com/jamiealquiza/envy v1.1.0
go: downloading github.com/abbot/go-http-auth v0.4.0
go: downloading github.com/disintegration/imaging v1.6.2
go: downloading github.com/korandiz/v4l v1.1.0
go: downloading github.com/pbnjay/pixfont v0.0.0-20200714042608-33b744692567
go: downloading nhooyr.io/websocket v1.8.10
go: downloading github.com/gen2brain/base64 v0.0.0-20221015184129-317a5c93030c
go: downloading golang.org/x/crypto v0.18.0
go: downloading golang.org/x/net v0.20.0
go: downloading golang.org/x/image v0.15.0
go: downloading github.com/spf13/cobra v1.8.0
go: downloading github.com/spf13/pflag v1.0.5
go: downloading golang.org/x/sys v0.16.0
$

$ ~/apps/cam2ip/bin/cam2ip  -index 1 -height 3840
Listening on :56000

```

The camera appears twice as /dev/video* files. That confuses cam2ip. I have an internal UVC camera and the external, so 4 `/dev/video*` files in total.

While the first device of a pair cannot be used by guvcview it must be used by cam2ip. But I only get 640x480 resulution.

```
$ v4l2-ctl --list-devices
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-11):
	/dev/video0
	/dev/video1
	/dev/media0

4K USB Camera: 4K USB Camera (usb-0000:00:14.0-2):
	/dev/video2
	/dev/video3
	/dev/media1

$ v4l2-ctl --list-formats-ext --device /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
[... 121 lines cut ...]
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
$
```

From another shell I can run:

```
$ mpv http://127.0.0.1:56000/mjpeg --fps=25 --fs
```

gen2brain, 2024-02-17:
I changed recently the default to be internal native jpeg, and not libturbo-jpeg. That is slower but I cannot handle the issues people who don't know how to compile are sending. Also, every cam has resolution, if the cam is 640x, you cannot "request" higher, you need to get better camera. Of course, you can do software scale.

mmokrejs, 2024-02-17:
@gen2brain Thank you for your answer. I read a bit the code because the README.md were not detailed enough to give me a clue how to handle the tool. I suggest showing more complicated examples:

```
# neither -u nor -y JPEG aka yuvj422p
# --uyvy or -u aka UYVY aka yuvj422p
# --yuv or -y aka YUYV aka yuv420p (my mobile app cannot read the MJPEG stream but mpv works fine)
mjpg_streamer -i "/usr/lib64/mjpg-streamer/lib64/input_uvc.so -d /dev/video2 -r 3840x2160 -f 25 -pl 50hz -q 98 -y" -o "/usr/lib64/mjpg-streamer/lib64/output_http.so -w /usr/share/mjpg-streamer/www -l 192.168.0.452"
mjpg_streamer -i "/usr/lib64/mjpg-streamer/lib64/input_uvc.so -d /dev/video4 -r 3840x2160 -f 25 -pl 50hz -q 98 -y" -o "/usr/lib64/mjpg-streamer/lib64/output_http.so -w /usr/share/mjpg-streamer/www -l 192.168.0.45 -p 8888"
```

Then I could access both instances with my camera 2 and 3 (camera 0 is laptops internal, yet another UVC):

```
http://192.168.0.45:8080/?action=stream
http://192.168.0.45:8888/?action=stream
```

I did not manage to use the `--fourcc RGB565` or `--fourcc RGB24` for some reason. Also not setup UDP streaming server or RTSP server.

gen2brain, 2024-02-17:
Thanks, but that all seems to be related to mjpg_streeamer app, right? I just heard about it. Readme should have just basic info imo, the rest is up to user.

mmokrejs, 2024-02-17:
Yes, this about mjpg_streeamer app, which allows me to select even the maximum resolution of my camera.

gen2brain, 2024-02-17:
If you can, please send a PR for readme, where  ffplay and vlc are, but just one example please.

closed 2025-06-15 by gen2brain