gen2brain/cam2ip #47: Timestamp Scaling?

incith, 2023-07-29:
Could we get a simple option to perhaps scale the timestamp size?  It's verrrry tiny on my 1080 stream!

Otherwise, this is great.  For an old iPhone (I'm using ivCam connected to a Windows PC) this is absolutely the way to go for Frigate NVR usage.  It's so smooth.  Appreciate what you've created!

gen2brain, 2023-07-30:
Sorry, currently it uses https://github.com/pbnjay/pixfont, which is a simple lightweight pixmap font without support for scaling (8x8 pixel fixed width). In order to add an option for scaling something like freetype is needed, probably with TrueType runtime font dependency etc. That will complicate and add complexity to simple tasks, not sure I want to do that.

incith, 2023-07-30:
Understood, thank you

closed 2023-07-30 by incith