gen2brain/cam2ip #48: fatal error: jpeglib.h: No such file or directory - Windows 10 - How to install libjpeg-turbo

toom94, 2023-11-08:
Hi,

I get an error when compiling on windows 10.

[image]

libjpeg-turbo installed
libjpeg-turbo-2.0.0-gcc.exe
path:
[image]

I have no idea what's wrong, please help

gen2brain, 2023-11-08:
Use `-tags jpeg` to build without CGO. Everything is in README, and you can search issues for many similar "issues".

closed 2023-11-08 by gen2brain

toom94, 2023-11-08:
--tags jpeg doesn't work for my camera. I need to build this with CV2. None of the previous issues solve my problem, please give me the correct instructions to install libjpeg-turbo on windows.

gen2brain, 2023-11-08:
Sorry, I have no idea how to install something on Windows. I cross-compile from Linux. You are looking for general instructions, i.e. for MinGW, for msys, whatever you are using, or just install it anywhere and point CGO_CFLAGS/CGO_LDFLAGS. Basic C knowledge about compilation is required.

gen2brain, 2023-11-08:
Btw. `-tags jpeg` has nothing to do with the camera, opencv or not. You can build with OpenCV and with `-tags jpeg`, i.e. native Go `image/jpeg` encoder.

toom94, 2023-11-08:
-tags jpeg returns:

socket: read: camera: can not retrieve frame: EOF
[image]

toom94, 2023-11-08:
mam zainstalowane opencv w wersji  opencv-2.4.13

mam libjpeg-turbo w wersji: libjpeg-turbo-2.0.4-vc64.exe 

mama ustawione parametry go env:
[image]

I have paths:
[image]

toom94, 2023-11-08:
ok, it worked, I built it with cv2 - but the application closes without any error... how can I find out the reason for closing?
[image]

gen2brain, 2023-11-08:
Start the terminal first, and run it from terminal, so it stays when it crashes.

toom94, 2023-11-08:
doesn't write anything
[image]

toom94, 2023-11-13:
I managed to get something more:
[image]

Have you encountered such a problem?