gen2brain/cam2ip #50: fails to build on some architectures..

alexmyczko, 2024-02-12:
see https://buildd.debian.org/status/package.php?p=cam2ip

gen2brain, 2024-02-12:
I am not sure how that  is an issue with this repo, what fails to build, and what the error is. I hope you don't expect me to go through all this. The requirements are clearly stated, so what this repo can do to help you?

closed 2024-02-13 by alexmyczko

gen2brain, 2024-02-14:
There is nothing `not planned` here, the screenshot of "something" is not an issue at all. So what is the real issue here?

alexmyczko, 2024-02-14:
nothing really, i have no idea of golang, so i do not know why ot failed to build on these architectures. thought maybe you might know..

gen2brain, 2024-02-14:
This project uses cgo, i.e. C libraries, so if you are to cross-compile, you need to know what you are doing. I think that is all, but if there is anything else I am willing to fix it.

alexmyczko, 2024-02-14:
i am not cross compiling, what happens at posted url is. the source is downloaded to a comouter of the architecture, and is being tried to be compiled… but really, it is not important, and *i* have no such architecture to run it on. i am fine with arm64 and amd64…

gen2brain, 2024-02-14:
The error mesage would be fine. I still have no idea what fails, and what is the actuall error and for what arch etc. You only sent some screenshot.

alexmyczko, 2024-02-14:
no i sent an url, this https://buildd.debian.org/status/package.php?p=cam2ip

and it says it can not build for ppc64, and a few more it failed. scrolling down all the red ones are not good. green is it built…

alexmyczko, 2024-02-14:
```
Tail of log for [cam2ip](https://buildd.debian.org/status/package.php?p=cam2ip&suite=sid) on [mips64el](https://buildd.debian.org/status/fetch.php?pkg=cam2ip&arch=mips64el&ver=1.6-2&stamp=1707749800&raw=0):

crypto/x509
github.com/spf13/pflag
mime/multipart
vendor/golang.org/x/net/http/httpguts
crypto/tls
github.com/spf13/cobra
github.com/jamiealquiza/envy
net/http/httptrace
net/http
github.com/abbot/go-http-auth
nhooyr.io/websocket
github.com/gen2brain/cam2ip/handlers
github.com/gen2brain/cam2ip/server
dh_auto_build: error: cd _build && go install -trimpath -v -p 4 github.com/gen2brain/cam2ip/cmd/cam2ip returned exit code 1
make: *** [debian/rules:7: binary-arch] Error 25

gen2brain, 2024-02-14:
Exactly, I have no idea what you are doing, and there is no error message, you are trying to build deb package right? How and why I do not care. Why not try to build cam2ip instead and send the error? I don't care about some deb errors, how is that related to this repo?

gen2brain, 2024-02-14:
@alexmyczko can you simple do a go build and paste errors here?