gen2brain/cam2ip #54: Changing resolution Windows

Pepepele2, 2024-04-04:
Hello,
I would ask for an instruction how to change resolution under Windows OS?

gen2brain, 2024-04-05:
Do `-width` and `-height` options from README work?

Pepepele2, 2024-04-05:
But where should I use it? The README file doeasn't explain a thing how to use this options with the .exe

gen2brain, 2024-04-05:
You pass it like this `cam2ip.exe -width 640` or `--width`, similar to how every app that has options works. It would be too much to create examples and docs for that.  Similar to how you can just set the `CAM2IP_WIDTH` environment variable, of course, I will not explain what is ENV, this is not the place for that.

Pepepele2, 2024-04-05:
But should i start it trough cmd or how to set this parameters?

gen2brain, 2024-04-05:
Yes, you start it from cmd with options, or you can set the `CAM2IP_WIDTH` and `CAM2IP_HEIGHT` environment variables.

Pepepele2, 2024-04-05:
Thank you very  much for the answer :)

closed 2024-04-05 by Pepepele2